Jaeyoung-Lim/mavros_controllers #188: Add Lissajous curves as reference trajectory

Jaeyoung-Lim, 2021-07-20:
**Problem Description**
Add a Lissajous reference trajectory so that the we can excite all axis of the vehicle for control inputs

The lissajous trajectory has been used for system identifcation for multirotors in a number of papers now:
- Böhm, Christoph, Martin Scheiber, and Stephan Weiss. "Filter-Based Online System-Parameter Estimation for Multicopter UAVs."

mzahana, 2021-09-13:
@Jaeyoung-Lim I am wondering how you intend to use this trajectory? Are you doing system identification?

Jaeyoung-Lim, 2021-09-13:
@mzahana Yes, this will be presented in the PX4 Dev summit

mzahana, 2021-09-13:
@Jaeyoung-Lim which session?

Jaeyoung-Lim, 2021-09-13:
@mzahana This would be included in this talk: https://px4summit2021.sched.com/event/m8VY/a-pipeline-for-data-driven-dynamics-modelling-for-uavs-using-flight-logs-manuel-galliker-eth-zurich