Jaeyoung-Lim/mavros_controllers #192: some questions about the tracking performance

958117216, 2021-08-30:
Hello, I have encountered some problems, I am very confused and hope to get your help.
I use the default parameters to run
roslaunch geometric_controller sitl_trajectory_track_circle.launch
 
[image]

The plane can't keep up with the target well.

so I turn down  the  angular velocity of target and rerun the program.
        <param name="shape_omega" value="0.5" />
[image]

Jaeyoung-Lim, 2021-08-30:
@958117216 What is your question?

958117216, 2021-08-31:
I turn down the angular velocity of target ,Shouldn't the tracker follow better? Why does the plane flip? As shown in Figure 2

958117216, 2021-08-31:
Is it a problem with the  firmware version? I use  v1.11.0-beta1 to simulate.

Jaeyoung-Lim, 2021-08-31:
@958117216 What did you tune down exactly? and why would turning down the angular velocity target result in better following performance?

and when does the vehicle flip?

958117216, 2021-09-01:
@Jaeyoung-Lim     Sorry for not being able to express clearly.

1.  I change the  angular velocity of the  circular trajectory, i.e. shape_omega.When the trajectory speed decreases, shouldn’t the trajectory tracking controller be easier to track?

2. I suspect it is a firmware version problem. Two versions v1.9.2 and v1.11.0-beta1 were selected for the simulation.Using different firmware, I recorded some videos

v1.9.2：
<param name="shape_omega" value="1.2" /> , that is default parameter. In this case, there is a gap between the quadrotor and the target in the x-y plane.

https://user-images.githubusercontent.com/46185326/131608040-742ec310-b508-4700-b070-00122720eb84.mp4

log:https://logs.px4.io/plot_app?log=fdd5f671-9658-49e9-a4f5-76d3f5a9f5b0

 <param name="shape_omega" value="0.5" /> In this case,the vehicle flipped.

https://user-images.githubusercontent.com/46185326/131608073-8b8d015d-e2cd-4695-b72e-272f14c80905.mp4

log:https://logs.px4.io/plot_app?log=f2f6f45f-fa73-4390-a99b-ab263873365d

v1.11.0-beta1:
Regardless of the value of "shape_omega", it is the following situation

https://user-images.githubusercontent.com/46185326/131608231-f0642998-0afe-4cce-a4b2-93e6e554c9fc.mp4

log:https://logs.px4.io/plot_app?log=c97c2ab4-b3e1-43ec-a1ca-05a6f97c010c

This looks weird, is my firmware compilation wrong? I tried it many times and it turned out to be like this.

Look forward to and thank you for your help.

closed 2021-12-24 by 958117216