Jaeyoung-Lim/mavros_controllers #194: Mavros_controllers not working with px4 v 1.11.0

shubham-shahh, 2021-09-30:
Hi, I am trying to use Mavros controllers with [this](https://github.com/HKPolyU-UAV/E2ES) repo and they are using px4 v 1.11.0 and I am unable to take off with mavros_controllers in the repo mentioned above

thanks

shubham-shahh, 2021-10-11:
Hi, I am trying to use [this](https://github.com/ZJU-FAST-Lab/ego-planner) planner along with [E2ES](https://github.com/HKPolyU-UAV/E2ES) simulation. I am passing the trajectory obtained from the planner to mavros controller and generating attitude controls and passing it to e2es simulation and the quad is not responding to the output of the mavros controllers

Jaeyoung-Lim, 2021-11-17:
@shubham-shahh I suspect that you are publishing setpoints to the wrong topics.

closed 2021-12-23 by Jaeyoung-Lim

shubham-shahh, 2021-12-23:
> @shubham-shahh I suspect that you are publishing setpoints to the wrong topics.

It was not related to topics, there were some version conflicts. I am sorry I didn't post a update. 

thanks