Jaeyoung-Lim/mavros_controllers #203: px4 inbuilt controllers and geometric controller

Ayush1285, 2021-12-27:
If we launch posix_sitl.launch then px4 will run its inbuilt control modules (that's mc_pos_ctrl,mc_att_ctrl,mc_rate_ctrl) and since we are running geometric controller node parallelly, therefore there will be two controllers running simultaneously, how you've sorted this issue?

Jaeyoung-Lim, 2021-12-27:
@Ayush1285 This repository is using the offboard interface and therefore is able to bypass the cascaded loop depending on what kind of setpoint you are sending. You can read more on how offboard mode works in https://docs.px4.io/master/en/flight_modes/offboard.html

Ayush1285, 2021-12-27:
Since you are sending Attitude Target setpoint that's why it can bypass position ctrl loops, right?
So, it is like, you're using the geometric controller for position control and then using cascaded attitude and rate controllers of px4 to achieve attitude target set by geometric control. 
Please correct me if I am wrong😅

Jaeyoung-Lim, 2021-12-28:
@Ayush1285 This package contains code for sending various setpoints such as position/velocity/acceleration or attitude rates(body rates). 

By sending attitude rates we are only using the rate controller of PX4

Ayush1285, 2021-12-28:
okay, thanks for the info.

closed 2021-12-28 by Ayush1285