Jaeyoung-Lim/mavros_controllers #205: Failsafe controller implementation

shubham-shahh, 2022-02-11:
Hi,
do you have any plans of implementing Failsafe controllers? which can handle landing without a position estimate?

thanks

shubham-shahh, 2022-02-17:
Hi, In case of a flight where the autopilot has no secondary sources to fallback, there is a need of a failsafe controller which can aid landing without a position estimate.
If we implement a failsafe controller in this package, mavros_controllers can be used as an end to end solution for vision based flights where there are no other sources to fallback

thanks

Jaeyoung-Lim, 2022-02-17:
> Hi, In case of a flight where the autopilot has no secondary sources to fallback, there is a need of a failsafe controller which can aid landing without a position estimate.

Not sure if I understand this correctly. The flight controller already has safety fallbacks and can safely land without position estimates. If we want a fallback policy it should be inside the flight controller and not on the companion computer since the flight controller is more reliable than the companion computer in general

shubham-shahh, 2022-02-17:
> Not sure if I understand this correctly. The flight controller already has safety fallbacks and can safely land without position estimates. If we want a fallback policy it should be inside the [...]

Hi, for example this [package](https://github.com/ctu-mrs/mrs_uav_controllers) implements a failsafe controller. If I am not wrong, without any external postion estimate, autopilot can only depend on dead reckoning based method for landing, and the drift  is accumulated, which results in a poor landing, please correct me if I am wrong or I'm missing out on something

Jaeyoung-Lim, 2022-02-27:
@shubham-shahh I think the dead reckoning part really doesn't change whether you have a failsafe controller outside the fmu. (which the fmu already has in the PX4 case)