Jaeyoung-Lim/mavros_controllers #206: Auto-tuning

brunopinto900, 2022-02-21:
Hello,

Do you think its possible to write a script to tune the geometric controller automatically with logs? The idea is to tune the controller gains for any quadrotor, update them on the respective config files. What do you think? I am looking for a self-contained project and this seems like a good idea.

Thank you.

Jaeyoung-Lim, 2022-02-21:
@brunopinto900 Sounds cool! And should be doable. If your interested, we already worked on a system identification pipeline through flight logs: https://github.com/ethz-asl/data-driven-dynamics

brunopinto900, 2022-02-21:
> @brunopinto900 Sounds cool! And should be doable. If your interested, we already worked on a system identification pipeline through flight logs: https://github.com/ethz-asl/data-driven-dynamics and you can take advantage of the infrastructure (log parsing etc)

Haha, found that work literally 5 minutes later.

Jaeyoung-Lim, 2022-02-21:
@brunopinto900 So lissajous trajectories are just a nice way to excite the system and this is what I used for  https://github.com/ethz-asl/data-driven-dynamics

Could be used for autotuning, but not necessarily required.

Jaeyoung-Lim, 2022-07-04:
Closing as stale

closed 2022-07-04 by Jaeyoung-Lim