Jaeyoung-Lim/mavros_controllers #207: More controllers

brunopinto900, 2022-02-21:
Sorry for opening another issue. Do you have any plans to add other controllers? If so, can you tell the reference papers? For example i was planning do add this https://foehnx.github.io/publication/icra18/ but its already done here https://github.com/llanesc/lqr-tracking. 

Thank you.

Jaeyoung-Lim, 2022-02-21:
@brunopinto900 Would be definitely cool to have that as part of this package, and we can then start doing benchmarks etc

Jaeyoung-Lim, 2022-02-21:
@brunopinto900 Personally, I would be interested on adding a predictive controller, e.g. a MPPI(Model Predictive Path Integral Controller) or a MPCC(Model Predictive Contouring Control) but probably a lot more work than a full state LQR :wink:

brunopinto900, 2022-02-21:
A model predictive control with collision avoidance would be nice. We can start by first implementing a linear MPC with no collision avoidance, then start integrate collision avoidance with RRT using OMPL (similar to PX4-Obstacle_avoidance repo). The MPPI is nice, but the math is so dull haha.

closed 2022-07-04 by Jaeyoung-Lim