Jaeyoung-Lim/mavros_controllers #213: Actualy Mine is not an issue. Just wondering

O2KAN, 2022-04-06:
Currently, I am doing a project. In that project, there are two-controller upper and lower. In lover control which will be px4 (in real system pixhawk). The upper controller will be Jetson nano or now my computer. In Jetson nano, I have some python codes that solve swarm behaviour. I have one problem now, I would like to send my desired positions that come from the swarm code which is python to the px4. Now before the real system application, I want to see the behavior and the results on gazebo using Ros/mavros sitl. While I was searching I found this repository of your mavros control codes. There are some launch files and c++ files for trajectory tracking. But what I want is not trajectory tracking. I want to do simultaneously send desired positions to drones using some bridge (this is my problem) to your repository to do the motions. How I can do that bridge? Can I use your repository for this purpose? Do you have any suggestions and tips for me? @Jaeyoung-Lim

Jaeyoung-Lim, 2022-04-06:
> I want to do simultaneously send desired positions to drones using some bridge (this is my problem) to your repository to do the motions.

@O2KAN Not sure what you mean by a bridge. This repo already has an example of multivehicle deployment. You can simply change the trajectory reference to a centralized trajectory publisher to control the vehicles in a coordinated way

O2KAN, 2022-04-06:
The bridge is something for connecting python code to ros not like ros bridge or whatever. I will try to do that after fixing some errors. Probably like you said publish the result of my python code and subscribe to them. How to make them the same frequency and simultaneously?

Jaeyoung-Lim, 2022-04-06:
@O2KAN You can publish at the same frequency using rospy

closed 2022-04-16 by O2KAN