Jaeyoung-Lim/mavros_controllers #215: need a new branch

shubham-shahh, 2022-04-21:
Hi @Jaeyoung-Lim,
I was thinking of pushing the changes required to make mavros_controllers work with ardupilot, if you can make a separate dedicated branch that would be awesome since there are a lot of changes I can make in the readme as well.

thanks and regards

Jaeyoung-Lim, 2022-04-21:
@shubham-shahh I would not want to create a separate branch just for ardupilot, since this would be hard to maintain. Since this package is interacting with mavros, the changes in the code required should be minimal.

If this is mainly for documentation, would it be better if we create a section in the readme for ardupilot? (or a wiki page)

shubham-shahh, 2022-04-21:
Hi, @Jaeyoung-Lim  thanks for the response, I can provide full support and maintenance required for the branch. I wanted a new branch so that we can have an independent implementation of fixes/features etc for the required firmware.

do let me know your thoughts on this

thanks

I'll try to add the solution to the current implementation

Jaeyoung-Lim, 2022-04-21:
@shubham-shahh I still think it is beneficial on both sides to have a single implementation that works for both.

IMHO, if you need an independent implementation just for your use case, there is no benefit of having a separate branch in the main repository, but a separate fork.

shubham-shahh, 2022-04-21:
@Jaeyoung-Lim, I understand your view, it is not specific to my use case and hence I am willing to add a PR to the main repo

keeping your concern in mind, I'll add the changes to the main repo and add the required documentation in the current readme in  a separate section

Thanks

closed 2022-04-21 by shubham-shahh

Jaeyoung-Lim, 2022-04-21:
@shubham-shahh Thanks!

halil93ibrahim, 2022-06-16:
Hi @shubham-shahh, I am having problems using the controller with Mavros/Ardupilot SITL. It seems you have solved issues with that. Do you have any particular version in this or your forked repo that works well with Ardupilot SITL?