Jaeyoung-Lim/mavros_controllers #217: roslaunch mavros_posix_sitl.launch WARN  [PreFlightCheck] Preflight Fail: Accel 1 inconsistent - Check Cal

erha527, 2022-05-19:
Hi~
    I am having issues running the  mavros_posix_sitl.launch on my px4-mavros-gazebo sitl. When I    `roslaunch mavros_posix_sitl.launch`    ,I meet an issue that __

WARN  [PreFlightCheck] Preflight Fail: Accel 1 inconsistent - Check Cal

so that  I can not arm the vehicle ,and I can also not use the qgc to arm.But when I  change the <arg name="vehicle" default="iris"/> to <arg name="vehicle" default="iris_rplidar"/>   ,it makes useful. I did not edit the iris.sdf,  and I need to use iris.sdf model.Can you help me?

Jaeyoung-Lim, 2022-05-19:
@erha527 `mavros_posix_sitl.launch` is not part of this repository. This is part of PX4: https://github.com/PX4/PX4-Autopilot/blob/master/launch/mavros_posix_sitl.launch

erha527, 2022-05-19:
> @erha527 `mavros_posix_sitl.launch` is not part of this repository. This is part of PX4: https://github.com/PX4/PX4-Autopilot/blob/master/launch/mavros_posix_sitl.launch

oh~sorry

> @erha527 `mavros_posix_sitl.launch` is not part of this repository. This is part of PX4: https://github.com/PX4/PX4-Autopilot/blob/master/launch/mavros_posix_sitl.launch

ohoh~sorry

closed 2022-05-27 by Jaeyoung-Lim