Jaeyoung-Lim/mavros_controllers #218: when i meet Error while "Running the code"

wahihijames, 2022-06-10:
I typing and start --> $ roslaunch geometric_controller sitl_trajectory_track_circle.launch

but I meet this error.
To solve the problem, what should I do?

(설치 다 끝내고, 실행 시키니깐, 아래와 같이 에러가 나와요) 
어디가 문제여서 발생한 걸까요?

================================================================================REQUIRED process [sitl-5] has died!

process has died [pid -1, exit code 255, cmd /home/wahihi/src/Firmware/build/px4_sitl_default/bin/px4 /home/wahihi/src/Firmware/build/px4_sitl_default/etc -s etc/init.d-posix/rcS __name:=sitl __log:=/home/wahihi/.ros/log/1271b0e8-e85c-11ec-9955-13fe95e71a1f/sitl-5.log].

log file: /home/wahihi/.ros/log/1271b0e8-e85c-11ec-9955-13fe95e71a1f/sitl-5*.log

Initiating shutdown!

================================================================================[ INFO] [1654827947.869619412, 3670.196000000]: Plugin gps_input initialized

[ INFO] [1654827947.871909011, 3670.200000000]: Plugin gps_rtk loaded

++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++

[ This is my log ]
아래는, 제가 실행 했을때, 터미널에서 나온 메시지 입니다.

wahihi@wahihi-VirtualBox:~/catkin_ws$ roslaunch geometric_controller sitl_trajectory_track_circle.launch

... logging to /home/wahihi/.ros/log/1271b0e8-e85c-11ec-9955-13fe95e71a1f/roslaunch-wahihi-VirtualBox-13686.log

Checking log directory for disk usage. This may take a while.

Press Ctrl-C to interrupt

Done checking log file disk usage. Usage is <1GB.

​

started roslaunch server http://wahihi-virtualbox:36183/

​

SUMMARY

========

​

CLEAR PARAMETERS

 * /mavros/

​

PARAMETERS

 * /gazebo/enable_ros_network: True

 * /geometric_controller/Kp_x: 10.0

 * /geometric_controller/Kp_y: 10.0

 * /geometric_controller/Kp_z: 20.0

 * /geometric_controller/Kv_x: 5.0

 * /geometric_controller/Kv_y: 5.0

[... 578 lines cut ...]
  File "<frozen importlib._bootstrap>", line 562, in module_from_spec

  File "<frozen importlib._bootstrap>", line 541, in _init_module_attrs

  File "<frozen importlib._bootstrap>", line 379, in cached

KeyboardInterrupt

    OPCODES = _makecodes("""

  File "/usr/lib/python3.8/sre_constants.py", line 69, in _makecodes

    items = [_NamedIntConstant(i, name) for i, name in enumerate(names)]

  File "/usr/lib/python3.8/sre_constants.py", line 69, in <listcomp>

    items = [_NamedIntConstant(i, name) for i, name in enumerate(names)]

  File "/usr/lib/python3.8/sre_constants.py", line 58, in __new__

    self = super(_NamedIntConstant, cls).__new__(cls, value)

KeyboardInterrupt

wahihi@wahihi-VirtualBox:~/catkin_ws$

Jaeyoung-Lim, 2022-06-10:
```
process has died [pid -1, exit code 255, cmd /home/wahihi/src/Firmware/build/px4_sitl_default/bin/px4 /home/wahihi/src/Firmware/build/px4_sitl_default/etc -s etc/init.d-posix/rcS __name:=sitl __log:=/home/wahihi/.ros/log/1271b0e8-e85c-11ec-9955-13fe95e71a1f/sitl-5.log].

log file: /home/wahihi/.ros/log/1271b0e8-e85c-11ec-9955-13fe95e71a1f/sitl-5*.log

Initiating shutdown!
```
Are you sure you have built PX4 SITL and run the sitl gazebo environment without using mavros?

wahihijames, 2022-06-10:
> ```
> process has died [pid -1, exit code 255, cmd /home/wahihi/src/Firmware/build/px4_sitl_default/bin/px4 /home/wahihi/src/Firmware/build/px4_sitl_default/etc -s etc/init.d-posix/rcS __name:=sitl [...]

Thank you Reply.
I solved my issue.
Although catkin_ws was already installed, it was installed again, and the path was tangled.

I have another question.
$roslaunch geometric_controller sitl_trajectory_track_circle.launch
When I run , all Gazebo etc. appear, but the drone is only arming and not taking off. (See attached video) What did I do wrong? The log file generated by QGroundControl is attached. please check.

[첨부파일 password: 1234](http://naver.me/5WGMhKHo)
#

catkin_ws가 이미 설치가 되어 있는데, 또 설치를 하게 되어서, path가 엉클어져서 생긴 문제 였습니다.
또 하나 질문이 있는데요. 
roslaunch geometric_controller sitl_trajectory_track_circle.launch
을 실행시키니, Gazebo 등은 전부 나오는데, 드론이 Arming 만 하고 있고, 이륙을 안하네요.  (첨부한 동영상 참조) 제가 무엇을 잘못해서 생긴것일까요?  QGroundControl에서 생성한 로그파일을 첨부 합니다. 확인 부탁드립니다.

Jaeyoung-Lim, 2022-06-10:
@wahihijames Was the problem solved?

On the video it looks like the video has already crashed to the ground

wahihi, 2022-06-10:
> On the video it looks like the video has already crashed to the ground

Thank you. your Reply. 
I solved this problem. Just Rebooting my Ubuntu..  It's likes crazy.   Ubuntu looks like Windows OS..  :)

Jaeyoung-Lim, 2022-06-10:
@wahihi Great to hear that it was fixed! Closing!

closed 2022-06-10 by Jaeyoung-Lim