Jaeyoung-Lim/mavros_controllers #221: roslaunch geometric_controller seems to be not working

nswang1994, 2022-06-28:
Here is the error output:

Resource not found: px4
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/ubuntu/catkin_ws/src/catkin_simple
ROS path [2]=/home/ubuntu/catkin_ws/src/mavros_controllers/controller_msgs
ROS path [3]=/home/ubuntu/catkin_ws/src/eigen_catkin
ROS path [4]=/home/ubuntu/catkin_ws/src/mavros_controllers/geometric_controller
ROS path [5]=/home/ubuntu/catkin_ws/src/mav_comm/mav_comm
ROS path [6]=/home/ubuntu/catkin_ws/src/mav_comm/mav_msgs
ROS path [7]=/home/ubuntu/catkin_ws/src/mav_comm/mav_planning_msgs
ROS path [8]=/home/ubuntu/catkin_ws/src/mav_comm/mav_state_machine_msgs
ROS path [9]=/home/ubuntu/catkin_ws/src/mav_comm/mav_system_msgs
ROS path [10]=/home/ubuntu/catkin_ws/src/mavros_controllers/mavros_controllers
ROS path [11]=/home/ubuntu/catkin_ws/src/mavros_controllers/trajectory_publisher
ROS path [12]=/opt/ros/melodic/share
The traceback for the exception was written to the log file

Jaeyoung-Lim, 2022-07-04:
@nswang1994 The readme explains how to setup your environment. I believe you have skipped this step: https://github.com/Jaeyoung-Lim/mavros_controllers#install-px4-sitlonly-to-simulate

nswang1994, 2022-07-04:
Thank you Jaeyoung. It works well now.

closed 2022-07-04 by nswang1994