Jaeyoung-Lim/mavros_controllers #222: implement to real world flights

ZhongmouLi, 2022-07-14:
Hi, Jaeyoung,

Thanks for your developed controller and it is really helpful to test algorithms.

It would be perfect if we can use this controller in real world flights. Do you have any plan to extend this controller for that purpose?

Jaeyoung-Lim, 2022-07-15:
@ZhongmouLi This package already works for real vehicles, and if you look into the closed issues numerous people have flown / are flying it on real systems

Is there anything missing on yourside to integrate this on a real system?

ZhongmouLi, 2022-07-19:
Hi, @Jaeyoung-Lim thanks for your reply.

I find your rep very user friendly for people in terms of drone simulation. It does save their time from setting environments and configurations.

For instance, I followed your instructions in readme to test controllers in gazebo simulation, which is well detailed. It motivates me to apply it on a real drone.

However, I find it harder to apply your controller into real experiments as no more guide is available now. I have to step into more details before I can apply your controller. PS: some code appear without comments and it takes more time.

Do you think it is of your interest to add some guides for apply this controller in real experiments?

Jaeyoung-Lim, 2022-07-20:
@ZhongmouLi Thanks for the suggestion

It is a lot harder to make a guideline in the manner you suggested, since everyhardware is different and there is no standard hardware setup that is commonly shared.

If you are interested in impro

Jaeyoung-Lim, 2022-08-04:
@ZhongmouLi Any updates?

ZhongmouLi, 2022-08-04:
Hi @Jaeyoung-Lim ,

I am applying mavros_controllers in real experiments. It is still under testing. I would like to contribute a guideline to implement  mavros_controllers with Pixhawk5 (PX4 1.13).

I will let you know you after my tests.

Jaeyoung-Lim, 2022-08-22:
@ZhongmouLi I have created a thread where people have shared videos of their system working in https://github.com/Jaeyoung-Lim/mavros_controllers/issues/225

Hope this helps

ZhongmouLi, 2022-08-22:
@Jaeyoung-Lim thanks for sharing this infor and it is very helpful to meet other people using mavros_controllers in real flights.

ZhongmouLi, 2023-02-10:
Hi @Jaeyoung-Lim ,

I am implementing mavros_controller in real experiments.

My drone cannot take off even it receives reference/setpoint and is in offboard. Could you please help me on that?

Now, I am sending a take-off trajectory to reference/setpoint that is subscribed by a geometric_controller node. The geometric_controller node receives pose information from Vicon.

However, my drone does not take off and it detects taking off but detects landing 2s after as you can see
https://review.px4.io/plot_app?log=6dbe24a5-d922-4e2f-ac49-b7e07a87fbf3
https://review.px4.io/plot_app?log=a86f69d5-fb48-4134-b569-43566c0caed9
https://review.px4.io/plot_app?log=5e60d515-66fa-4d72-af49-37a9df34f74f
https://review.px4.io/plot_app?log=8a3daaf5-a213-4d28-b42d-0320ce4227d2.

Thank you.

Jaeyoung-Lim, 2023-02-10:
@ZhongmouLi As you can see in the logs, the position setpoints are either not reaching the vehicle, or not being logged.

Are you sure that the references are being received by the autopilot?

ZhongmouLi, 2023-02-14:
@Jaeyoung-Lim Thanks for your reply.

A ref trajectory is sent to reference/setpoint, based on which the geometric_controller node computes control inputs and send thrust and body rates to /mavros/setpoint_raw/attitude.

Do you think the drone is swtiched back to landing, because the commands sent to /mavros/setpoint_raw/attitude are not well received?

Jaeyoung-Lim, 2023-04-26:
@ZhongmouLi Awesome! Was there a part that you had to trouble shoot the system? Or did it work out of the box?

Videos wpuld be aplreciated :)

Jaeyoung-Lim, 2023-04-26:
>  I have done several modifications to transfer the package from simulation into experiments.
@ZhongmouLi What did you have to do? I am curious because I currently fly the package on a real vehicle. So in principle there should be no modification needed.

fangjir, 2023-07-12:
hi,@Jaeyoung-Lim 
i have a question in replying the package.What I learned before was that when entering offboard mode, a position signal with a frequency greater than 2Hz needs to be sent. However, in the program, off board is performed in a timer that occurs every 1s。Does it mean that in the package, cmdloop is sending signals before offboardmode

Jaeyoung-Lim, 2023-07-12:
@fangjir Yes, you need to send setpoints before neing able to switch to offboard mode

ZhongmouLi, 2023-11-26:
I control a drone by sending it thrust+attitude in order to follow a reference trajectory.

Here is what I have modified:

1. add a trajectory lib to generate a reference trajectory to enable the drone to take off from where it is  up to, for example, 1m, before following a real reference trajectory. 

Because as you know, some reference trajectory before, made the drone take off aggressively or just directly following the reference trajectory even the drone is on the ground. I believe a soft taking off is better for beginners.

2. I add a trajectory planner that sends a desired position that is the home position to the drone to prepare for taking off. But the rest reference trajectory will only be sent once the planner finds the drone is in the offboard mode. 

This is because, I found in experiments, the drone sometimes does not switched to the offboard mode yet, while the planner sends desired positions that are 1m in the air. Then the drone takes off badly.

Generally speaking, my modification is to make a soft take off.

> > I have done several modifications to transfer the package from simulation into experiments.
> 
> @ZhongmouLi What did you have to do? I am curious because I currently fly the package on a real vehicle. So in principle there should be no modification needed.
> 
> Plugging vicon into the system is a slightly different story, since this is more related to the state estimate and not necessarily with the controller itself.

ZhongmouLi, 2024-04-15:
Hi @Jaeyoung-Lim , I would like to share the video of flying a quadrotor using mavros_controller. Hope it can encourage others to use it.

https://youtu.be/WpRVAdvC_cw?si=kMHjiEhdHg0yH8SN

Thanks again for your help on mavros_controller.

Jaeyoung-Lim, 2024-04-16:
@ZhongmouLi Awesome! Thanks for sharing!