Jaeyoung-Lim/mavros_controllers #228: [question] Math behind converting attitude errors to body rates

aihamalharbat, 2022-10-21:
Hi @Jaeyoung-Lim ,

I have a couple of questions:
1. What is the logic/math behind converting the non-linear attitude errors to angular velocity setpoints (as shown below)? is there a physical meaning to this conversion and to the constant `attctrl_tau_`?
https://github.com/Jaeyoung-Lim/mavros_controllers/blob/8b3fff0327b56c415aa24708bea5f37d76307404/geometric_controller/src/nonlinear_geometric_control.cpp#L63
and
https://github.com/Jaeyoung-Lim/mavros_controllers/blob/8b3fff0327b56c415aa24708bea5f37d76307404/geometric_controller/src/nonlinear_attitude_control.cpp#L56-L58

2. Do you have any practical recommendations on how to get an accurate and proper normalization of the thrust command, i.e. how to estimate the `norm_thrust_const_` and `norm_thrust_offset_`? 
https://github.com/Jaeyoung-Lim/mavros_controllers/blob/8b3fff0327b56c415aa24708bea5f37d76307404/geometric_controller/src/geometric_controller.cpp#L408-L410

Thanks in advance!

cillian-bao, 2023-01-10:
I encounter the same problem，if the code written in this way, I dont't think it is a geometric control approach. @Jaeyoung-Lim

Jaeyoung-Lim, 2023-01-10:
@cillian-bao I have explained why this is the case in https://github.com/Jaeyoung-Lim/mavros_controllers/issues/230