Jaeyoung-Lim/mavros_controllers #229: Tune for PX4

Cavalletta98, 2022-11-11:
Hi, which are the parameters for a drone with PX4 fw onboard? I'm using version 1.13.1 of the fw
Thanks

Jaeyoung-Lim, 2022-11-11:
@Cavalletta98 Not sure if I follow your question.

You need to tune your vehicle and the guidelines can be found in https://docs.px4.io/main/en/config_mc/pid_tuning_guide_multicopter.html

Cavalletta98, 2022-11-11:
I will explain what i'm doing. I'm combining the [Fast Planner](https://github.com/HKUST-Aerial-Robotics/Fast-Planner) with PX4 fw in simulation (ROS + gazebo). To do so, i'm using this controller to send the trajectory to the drone. To send the trajectory from Fast Planner to the controller, i'm using this [code](https://github.com/mzahana/px4_fast_planner/blob/master/scripts/trajectory_msg_converter.py).
The problem is that the drone is jerking at the beginning (before a send a goal). I think the problem could be the parameters of the controller.

I attached the parameters:

`<arg name="mav_name" default="iris"/>
  <arg name="command_input" default="2" />
  <arg name="gazebo_simulation" default="true" />
  <arg name="max_acc" default="5.0" />
  <arg name="init_pos_x" default="0.0" />
  <arg name="init_pos_y" default="0.0" />
  <arg name="init_pos_z" default="2.0" />
  
  <node pkg="geometric_controller" type="geometric_controller_node" name="geometric_controller" output="screen">
  		<param name="mav_name" type="string" value="$(arg mav_name)" />
          <remap from="command/bodyrate_command" to="/mavros/setpoint_raw/attitude"/>
          <param name="ctrl_mode" value="$(arg command_input)" />
          <param name="enable_sim" value="$(arg gazebo_simulation)" />
          <param name="enable_gazebo_state" value="true"/>
          <param name="velocity_yaw" value="false"/>
          <param name="max_acc" value="$(arg max_acc)" />
          <param name="Kp_x" value="6.0" />
          <param name="Kp_y" value="6.0" />
          <param name="Kp_z" value="10.0" />
          <param name="Kv_x" value="1.5" />
          <param name="Kv_y" value="1.5" />
          <param name="Kv_z" value="3.5" />
          <param name="init_pos_x" value="$(arg init_pos_x)" />
          <param name="init_pos_y" value="$(arg init_pos_y)" />
          <param name="init_pos_z" value="$(arg init_pos_z)" />
  </node>`

Cavalletta98, 2022-11-11:
<arg name="mav_name" default="iris"/>
  <arg name="command_input" default="2" />
  <arg name="gazebo_simulation" default="true" />
  <arg name="max_acc" default="5.0" />
  <arg name="init_pos_x" default="0.0" />
  <arg name="init_pos_y" default="0.0" />
  <arg name="init_pos_z" default="2.0" />
  
  <node pkg="geometric_controller" type="geometric_controller_node" name="geometric_controller" output="screen">
  		<param name="mav_name" type="string" value="$(arg mav_name)" />
          <remap from="command/bodyrate_command" to="/mavros/setpoint_raw/attitude"/>
          <param name="ctrl_mode" value="$(arg command_input)" />
          <param name="enable_sim" value="$(arg gazebo_simulation)" />
          <param name="enable_gazebo_state" value="true"/>
          <param name="velocity_yaw" value="false"/>
          <param name="max_acc" value="$(arg max_acc)" />
          <param name="Kp_x" value="6.0" />
          <param name="Kp_y" value="6.0" />
          <param name="Kp_z" value="10.0" />
          <param name="Kv_x" value="1.5" />
          <param name="Kv_y" value="1.5" />
          <param name="Kv_z" value="3.5" />
          <param name="init_pos_x" value="$(arg init_pos_x)" />
          <param name="init_pos_y" value="$(arg init_pos_y)" />
          <param name="init_pos_z" value="$(arg init_pos_z)" />
  </node>

Jaeyoung-Lim, 2022-11-11:
@Cavalletta98 This implementation sends body rate commands to the vehicle. Therefore, you need to tune the rate controller of the vehicle before using this package. I have posted the link which explains how to do this in the previous answer

Cavalletta98, 2022-11-11:
Can you please post here a link of this tuning "guide",pls?

Jaeyoung-Lim, 2022-11-11:
@Cavalletta98 I already posted the link, in my first answer to your question

Cavalletta98, 2022-11-14:
I'm trying to use it but it is not so easy to do it. According to you, do you know a "better" way to use the trajectory of the Fast Planner with px4 fw?