Jaeyoung-Lim/mavros_controllers #230: why angular rate are sent?

fangjir, 2023-01-06:
Dear [Jaeyoung-Lim](https://github.com/Jaeyoung-Lim):
I have a couple of questions about geometric control.I have seen your  paper of “geometric tracking control  of a quadrotor UAV on SE(3)”. But after seeing the code, In your comment, I see this sentence, the moment is designed in the article, but in offboard mode, the angular rate command is sent. This is exactly what I am confused about. Since the angular rate is published, where does the torque designed in the article apply.
 
[image](https://user-images.githubusercontent.com/71303114/211003632-17404f95-4909-4bb9-a921-237113e08da9.png)

In addition, in the following two lines of code, the first line is the attitude error in the article, and the second line is exactly where I am confused.
[image]

Jaeyoung-Lim, 2023-01-06:
@fangjir The angular velocity is tracked by the autopilot. Therefore in this package, we are just using the attitude error metric from the paper

fangjir, 2023-01-06:
@Jaeyoung-Lim ，I'm so confused that why angular velocity are used instead of directly moment or angular acceleration (because M=J*alpha)。and Where is the torque formula designed in the paper used in practice

Jaeyoung-Lim, 2023-01-06:
@fangjir This is a ros package, which normally runs on a mission computer. Since the mission computer is connected via a serial interface to the autopilot controlling the torque via the serial interface results in large time delays and jittery controls.

That is why I have modified the controller to just use the error metric of the attitude error and use that error to command angular rate reference commands, so the rate control loop on the autopilot can track the controlled angular rate command

cillian-bao, 2023-02-16:
> 

hello, did you mean "J=M*alpha", alpha is the angular acceleration?