Jaeyoung-Lim/mavros_controllers #231: Migration to ROS2

mzahana, 2023-02-02:
Hi @Jaeyoung-Lim 

Any plans to make a version of this package for ROS2?

Jaeyoung-Lim, 2023-02-02:
@mzahana What parts would you want on ros2?

mzahana, 2023-02-02:
I would imagine refactoring the geometric controller node, and any related nodes, to subscribe to ROS2 topics to get reference trajectory, and send setpoint to PX4.

Jaeyoung-Lim, 2023-08-08:
@mzahana Any updates regarding this matter? I think it would take me max. 1 day to port this over if it is useful for anyone

mzahana, 2023-08-08:
It is definitely in the plan. I just have not had the time to work on it. Maybe I will get some time in a week or 2.

Jaeyoung-Lim, 2023-08-08:
@mzahana We can try to work together to also get it done more quickly.

I have most parts without the controller done here: https://github.com/Jaeyoung-Lim/px4-offboard

mzahana, 2023-08-25:
I saw your repo for the offboard controller in ros2.

Are you thinking of a c++ implementation or Python one ?
You use the Eigen library, and we would need to go C++ to have minimum migration time, compared to Python.

Also, I have a concern about the name of the package. If it is mostly a geometric controller, and mavros is not yet mature for ros2 integration, why would it be called mavros_controllers ?

I would explicitly call it something like geometric_controller_ros2, or something similar.

ZhongmouLi, 2025-06-16:
I have finished a draft for humble and PX4 1.15.4. I am testing it in Gazebo now.

ZhongmouLi, 2025-06-19:
I am having one question about mapping thrust into throttle after reading  [issues/185](https://github.com/Jaeyoung-Lim/mavros_controllers/issues/185), [issues/143](https://github.com/Jaeyoung-Lim/mavros_controllers/issues/143) and [issues/156](https://github.com/Jaeyoung-Lim/mavros_controllers/issues/156).

In PX4 1.15.4 using gz_x500,  I got the throttle, like 0.65, in Mavlink messages using Qgroundcontrol when it is hovering. May I ask how to use this value to compute ```normalizedthrust_constant``` and ```normalizedthrust_offset```.