Jaeyoung-Lim/mavros_controllers #232: Path Following with this geometric control

maulanaazhari, 2023-04-14:
Hi!
I am wondering if this geometric controller can track well if we don't have the target velocity or acceleration (in this case just a path). My use case is that I am planning to use path planner similar to A* which usually just provide the path to track. 

I know that the performance maybe reduced due to the missing information from the trajectory. But compared to PID position controller, what do you think will work better?

Thanks~

Jaeyoung-Lim, 2023-04-14:
@maulanaazhari You can still do path following control with the geometric controls. You just need a way to generate acceleration from the path geometry

maulanaazhari, 2023-04-14:
I see, do you have any reference on how to generate velocity and acceleration from the path geometry? do you think method like min-snap trajectory generator would be good?

Jaeyoung-Lim, 2023-04-14:
@maulanaazhari No, if you use a trajectory generator, you are tracking trajectory again, not a path

https://www.sciencedirect.com/science/article/pii/S1474667016419853

maulanaazhari, 2023-04-14:
got it! I am closing this comment, thanks!