Jaeyoung-Lim/mavros_controllers #238: Question about EKF Origin Setting for Offboard Flight Mode Transition

Mitchell-Lee-93, 2023-11-17:
Hi Jaeyoung,

I want to say that you've done some amazing work with this code! I've been exploring it for an indoor offboard flight in a real drone.

According to the PX4 documentation ([Link](https://docs.px4.io/main/en/ros/external_position_estimation.html)), even though local position data is obtained from Vicon, setting the EKF origin is necessary to establish a global position, a requirement for transitioning the drone into offboard mode. While going through the code,  however, it seems like there isn't a part in the code that sends the EKF origin information to PX4.

Could you please clarify how the transition to offboard mode is achieved without the global coordinate being explicitly sent? I might be missing something in the code, or misunderstood the transition to offboard mode.

Thanks in advance for your help and the great work on this project!

Jaeyoung-Lim, 2023-11-28:
@Mitchell-Lee-93 Sorry for the late response,

> Could you please clarify how the transition to offboard mode is achieved without the global coordinate being explicitly sent? I might be missing something in the code, or misunderstood the transition to offboard mode.

Offboard mode does not need any reference of global position. I think the confusion comes from the fact that auto flight modes (which do not include offboard) are tracking GPS waypoints.

Could you elaborate on why you are trying to use global position estimates

Mitchell-Lee-93, 2023-12-02:
Thank you very much for the clarification Jaeyoung!

To provide more detail, I successfully conducted indoor flights using an [Omnicopter](https://docs.px4.io/main/en/frames_multicopter/omnicopter.html). Following this, I attempted to use the [MAVROS C++ offboard control example](https://docs.px4.io/main/en/ros/mavros_offboard_cpp.html) to enable the Omnicopter to follow a predefined trajectory. However, when I tried to switch to offboard mode using the RC stick, I encountered the message, "Switching to mode 'Offboard' currently not possible." Initially, I thought this issue was due to the lack of a global position origin, but now I realize that might not be the case. 🤔

Jaeyoung-Lim, 2023-12-02:
> To provide more detail, I successfully conducted indoor flights using an [Omnicopter](https://docs.px4.io/main/en/frames_multicopter/omnicopter.html).

@Mitchell-Lee-93 Do you have a video? :)

> I thought this issue was due to the lack of a global position origin, but now I realize that might not be the case. 🤔

It means that you either don't have a local position estimate, or the offboard control message flags are not sent properly.

Mitchell-Lee-93, 2023-12-02:
I don't have a video of the experiment at the moment, but I'll be sure to share it here as soon as I have one.

Thank you very much for highlighting the potential issues. I'm confident that the system has a valid vision_pose and local position estimate, as it was able to fly in position control mode. Based on your suggestion, I will take a closer look at the offboard control message flags!

Jaeyoung-Lim, 2023-12-02:
@Mitchell-Lee-93 If you can share a flight log, would be easier to find what the problem is

Mitchell-Lee-93, 2023-12-11:
Thank you a lot for your help and advice! I would have kept wasting my time on the global position thing if you haven't pointed it out. I was able to make it fly offboard mode by changing the setpoint topic from '/mavros/setpoint_position/local' to '/mavros/setpoint_raw/local' in [MAVROS C++ offboard control example](https://docs.px4.io/main/en/ros/mavros_offboard_cpp.html). This is the flight video of 8-figure trajectory tracking while rotating, to check 6 DOF flight capability [Video](https://youtu.be/hq9_HEeeI5U). Thank you again for the help!

Jaeyoung-Lim, 2023-12-11:
@Mitchell-Lee-93 Thank you for the video!

I also have a ros2 implementation specifically for omnidirectional vehicles: https://github.com/Jaeyoung-Lim/px4-manipulation

Let me know if there is any way we can further collaborate :)

Mitchell-Lee-93, 2023-12-12:
That sounds awesome! I'll contact you by email :)

closed 2023-12-12 by Mitchell-Lee-93