Jaeyoung-Lim/mavros_controllers #239: Changing battery affects the tracking, and needs retuning parameters

anandvgeorge, 2023-12-04:
Hi!

I have been using this controller for a while in simulation and it works really nicely. Recently we started to use it on a real drone. We were able to tune the parameters, and the drone is tracking the inputs. The problem we face is, whenever we change the battery (using the same 3 cell type battery), these tuned parameters are not good any more. The drone seems to have more thrust initially and it does not track the inputs. There are always some overshoots in X-Y and steady state error in Z axis. By the time we tune parameters and do some tests, it's time to use another battery, which means another parameter tuning session. 

Have anyone had this problem? Or what could be the possible reasons for this?

Thanks for any insights!

Jaeyoung-Lim, 2023-12-05:
@anandvgeorge 

This is a common problem when you are using 3-phase controlled ESCs to control the motors without any feedback. Since we are controlling the normalized thrust values of the vehicle, this is quite a normal problem. The maximum thrust of a vehicle is defined by the voltage of the battery.

However, the tuning on attitude shouldn't be that bad since PX4 uses a cascaded control structure. You might want to check the integrator gains to make it robust against environmental disturbances(such as battery voltage drops).

anandvgeorge, 2023-12-05:
Thanks for the reply @Jaeyoung-Lim 

We enabled the parameter `MC_BAT_SCALE_EN` to scale thrust based on the battery voltage, but it didn't seem like the situation was improved. Would it be wise to do this scaling outside PX4 based on battery voltage and send the scaled value to MAVROS?

We will also try fine tuning the integrator gains and see the changes.

Jaeyoung-Lim, 2023-12-05:
@anandvgeorge I believe `MC_BAT_SCALE_EN` only works if you have feedback from the battery.

> Would it be wise to do this scaling outside PX4 based on battery voltage and send the scaled value to MAVROS?

You should do it if you want consistent acceleration control.

anandvgeorge, 2023-12-05:
Thank you. I will check these and update here.

anandvgeorge, 2023-12-13:
Hi. I have added another node to subscribe to the thrust value and to publish it after scaling based on the battery voltage, based on the initial mavros implementation by [this commit](https://github.com/PX4/PX4-Autopilot/commit/a634c14582a57ef50dadd281ba5816654eaf0aa9#diff-6db82b3153c7d34d4e343c32f9ec6004f340ed68b683774a5fcdd76d5e84eb69R227-R236). Now we don't see the thrust variation any more.

closed 2023-12-13 by anandvgeorge