Jaeyoung-Lim/mavros_controllers #243: No such file or directory  #include <controller_msgs/FlatTarget.h>

sniper006006, 2024-07-25:
[image]
Errors     << geometric_controller:make /home/auto_px4/autoware_px4_docker0715/catkin_ws/logs/geometric_controller/build.make.005.log                                      
In file included from /home/auto_px4/autoware_px4_docker0715/catkin_ws/src/mavros_controllers/geometric_controller/src/geometric_controller.cpp:41:0:
/home/auto_px4/autoware_px4_docker0715/catkin_ws/src/mavros_controllers/geometric_controller/include/geometric_controller/geometric_controller.h:66:10: fatal error: controller_msgs/FlatTarget.h: No such file or directory
 #include <controller_msgs/FlatTarget.h>

“Please advise, how can I solve this compilation problem? Thank you.”

Jaeyoung-Lim, 2024-07-25:
@sniper006006 Please share the full log. You probably are building with the wrong command.

leddat, 2024-09-19:
> ![img_v3_02d4_bda8832a-0cd3-40e2-9228-6a37f66045fg](https://private-user-images.githubusercontent.com/168155221/352058938-24f439eb-23ce-49d6-ba92-4a9986acb61f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY3MjkwODksIm5iZiI6MTcyNjcyODc4OSwicGF0aCI6Ii8xNjgxNTUyMjEvMzUyMDU4OTM4LTI0ZjQzOWViLTIzY2UtNDlkNi1iYTkyLTRhOTk4NmFjYjYxZi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTE5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkxOVQwNjUzMDlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01ZmJmZmJmYTkyNWUzM2ZmNzg4YjdmZThkZjRjNDFlM2RjZWZmMWJkZmNiMTBhNzA2ZjY1YWIxOGY5MGI4NzIwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.wo6Gfo3WvGhqhuACBV-YvXHyKlAnIu43GOQVEdAm1gY) Errors << geometric_controller:make /home/auto_px4/autoware_px4_docker0715/catkin_ws/logs/geometric_controller/build.make.005.log In file included from /home/auto_px4/autoware_px4_docker0715/catkin_ws/src/mavros_controllers/geometric_controller/src/geometric_controller.cpp:41:0: /home/auto_px4/autoware_px4_docker0715/catkin_ws/src/mavros_controllers/geometric_controller/include/geometric_controller/geometric_controller.h:66:10: fatal error: controller_msgs/FlatTarget.h: No such file or directory #include <controller_msgs/FlatTarget.h>
> 
> “Please advise, how can I solve this compilation problem? Thank you.”

Add controller_msgs in your CMakeLists.txt may help
```makefile
find_package(catkin REQUIRED COMPONENTS
  roscpp
  rospy
  dynamic_reconfigure
  tf
  mavros
  mavros_extras
  mavros_msgs
  mavlink
  controller_msgs
)

catkin_package(
  INCLUDE_DIRS include
  LIBRARIES geometric_controller
  CATKIN_DEPENDS roscpp rospy std_msgs mavros_msgs geometry_msgs sensor_msgs tf controller_msgs
)
```

Jaeyoung-Lim, 2024-09-19:
@leddat Do you mind creating a PR for the fix?

leddat, 2024-09-19:
> @leddat Do you mind creating a PR for the fix?

@Jaeyoung-Lim @sniper006006 probably forgot to run the command `catkin config --merge-devel` before build so I don't think we need to create PR for this