Jaeyoung-Lim/mavros_controllers #245: Tracking performance degradation on latest PX4 version (1.15)

heleidsn, 2025-02-16:
Hi @Jaeyoung-Lim,

Recently I have tested mavros_controller on the latest PX4 (v1.15), the tracking performance is not good comparing to the former verion such as v1.13. Attached are videos for 1.13 and 1.15. 

[V1.13](https://www.youtube.com/watch?v=UiYxzJd60B4&ab_channel=LeiHe) which is stable
[V1.15](https://www.youtube.com/watch?v=NAzEb06WYFI&ab_channel=LeiHe)

The only difference is the PX4 version. Any idea with the performance difference? 
Is there any other node to achieve trajectory tracking on PX4 on 2025? 

Thanks.

Jaeyoung-Lim, 2025-02-16:
@heleidsn Have you tried to retune the controller? I dont think the instability is coming from the implementation, given that we are sending body rate commands

heleidsn, 2025-02-16:
Hi @Jaeyoung-Lim, thanks for your reply. The tracking performs better after auto-tuning the PX4 rate controller using QGC. Currently, there may be some modifications on the controller or iris model. Will do more comparisons on the PX4 side to see if I can find some clues. 

1.13 with default parameters
[image]

1.15 with default parameters
[image]

1.15 after tunning
[image]

Parameters after tunning. 
[image]

heleidsn, 2025-02-16:
The yaw rate control loop is the key to improving the tracking performance, especially increasing the P and D terms.

heleidsn, 2025-02-16:
I checked with the [iris model history](https://github.com/PX4/PX4-SITL_gazebo-classic/commits/fbd8e9e6bbd2188de81677494f15885dead99c48/models/iris), there are not too many modifications. 

The most likely cause of tracking instability is the newly [added yaw torque low-pass filter in PX4](https://github.com/PX4/PX4-Autopilot/commit/8ecb76aba22a54e803edfedb9145290b92b3544f). The default cutoff frequency, `MC_YAW_TQ_CUTOFF`, is currently set to 2Hz, which is too low for agile flight. Changing it to 10Hz can resolve the tracking issue. 

I will close this issue as it relates to a controller update in PX4.

closed 2025-02-16 by heleidsn