Jaeyoung-Lim/mavros_controllers #247: how to tune the parameters

kaoxirui, 2025-03-28:
The angular velocity and pitch angle tracking of the drone are fine, but the roll will have a large rough adjustment. How can I adjust the parameters?

[image]

[image]

[image]

[image]

[image]

Jaeyoung-Lim, 2025-03-28:
@kaoxirui Please provide more information. Whar are the logs? Is it from a real vehicle? What are you trying to tune?

kaoxirui, 2025-03-28:
it's in simulation.I am wondering what csn i do to make the mavros/local_position/pose/pose/orientation/roll approach to  /reference/pose/pose/orientation/roll

kaoxirui, 2025-03-28:
@Jaeyoung-Lim it's in simulation.I am wondering what csn i do to make the mavros/local_position/pose/pose/orientation/roll approach to /reference/pose/pose/orientation/roll